ArduPilot/MissionPlanner #1678: Mission planner fails opening vlc streams

OlivierJB, 2017-10-30:
#### Issue details
Vlc is installed stock, and opens a stream fine. 
See below for error message. An additional windows trace confirmed dll was opened.
[image]

#### Version
1.3.50

#### Platform
Via control-F
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2017-11-04:
if you are on a 64bit os, you need 64bit vlc installed, and 32bit uninstalled

meee1, 2017-11-06:
current beta MP will try detect the skyviper and start the video automatically using gstreamer. you need gstreamer 1.9.2 installed first though, with the full install

closed 2017-11-06 by meee1

Sas-LAB, 2017-11-10:
**@meee1 Worked out for me !!! thanks!** 
(just 64bit vlc installed, and 32bit uninstalled)