ArduPilot/MissionPlanner #1679: Auto WP -> Survey grid layout error in complex shapes

suraqis, 2017-11-06:
#### Issue details
When Auto WP Survey Grid is built it fails on complex polygon shapes (with concavities). Image attached 
[image]

May it be a good solution - just ignore concavities when building layout and accept concavities when turning shutter control? Often it's takes less time to fly over all concavities but it takes a lot of space to keep photo from not interested area.

#### Version
Mission Planner 1.3.50

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Fixed Wing_

#### Hardware type
All

#### Logs
No need

meee1, 2017-11-06:
please provide the grid file use to generate this grid.

in the grid interface press control-s

anestetico, 2018-08-22:
The problem is still there but only affects users doing very large areas, anyway just a simple polygon like attached shows the problem, sometimes you can resolve by tweaking the angle, but not in complex polygons.
Thank you
[image]
[test.zip](https://github.com/ArduPilot/MissionPlanner/files/2312014/test.zip)

anestetico, 2018-09-05:
[image]

Naterater, 2018-12-02:
10/10 Fixed! Thank you so much!!!

[image]

anestetico, 2018-12-02:
> 10/10 Fixed! Thank you so much!!!
> 
> [image]

Look at your pictures, there are still strange paths, I downloaded latest beta and it seems better but not 100%
[image]

meee1, 2018-12-02:
you will never get 100% because of the shape you are flying. there will always be lines that will need to navigate to without being part of the grid. if you where to change the angle, you could possible find a more suitable flight line that had less transit paths.

anestetico, 2018-12-02:
Qgroundcontrol acts different, maybe sometimes is better and sometimes not, btw thank you very much for the fix, it will help a lot.
For the Corridor function would be possible to add corridor option and corridor width when saving *.grid file? If you load a previously saved file you have to remember that is a corridor (not difficult) and its width.

Naterater, 2018-12-03:
The "strange paths" are where it goes to finish sections. I don't consider them strange although I would prefer if it would turn the camera off on those paths. Super-crazy shapes with a million vertices are a very rare corner case, however L-Shaped fields are not.

Now that we are continuing the discussion, I would like to continue this improvement. 

#1 - Keep as many turns in the same direction as possible (ie - Into the wind). For the example above, consider the wind coming from the southwest. Mapping with a crosswind and turning into the wind is ideal. Unfortunately the eastern section will always have tailwinds during the turns causing an overshoot of the flight lines. Simple solution is to have waypoint "40" come after "28". That results in only one downwind turn instead of 3 (or more depending on the size of the grid).

[image]

#2 - Add overshoot without adding a million extra waypoints. Missions are best kept simple. Straight-through waypoints do nothing and only clutter the mission. They're only useful if "breakup starts" is on.

[image]

#3 - Break apart lead-in just like overshoot is broken up. Square corners can be turned well with the current navigation controllers. Acute ones often get ugly.

[image]

#4 - Fix the waypoint numbering in the survey (grid) window. Even without breakup starts, the waypoint numbers are incremented by two in the window. Once the waypoints are on the mission, they are incremented by 1.

[image]

meee1, 2018-12-03:
1. this assumption assumes you are using plane. I do see your point however. (whats im really saying is there are different rules depending on the use case)
2. the extra points are needed depending on what mode you use (servo/digicam). The reason they are useful is it causes the navigation controller to pull tighter to the start of the survey area,, setting up the line the plane will have at the entry of the survey area. ie there is a more asserted effort to get online as the cross track error is higher, vs having a wp at the far end. it comes down to how the l1 controller is tuned.
3.  sounds doable.
4. the number on this screen has never been accurate. turn on internals and you will see.

anestetico, 2018-12-06:
This is the tomorrow survey and the difference between MP and Qgroundcontrol.
MP 1.3.61 build 1.3.6913.26037 (this is the best plan after tweaking the angle to 178
Qgroundcontrol 3.4.4 this is the first plan leaving 0 degrees angle
[image]
[image]
[image]

meee1, 2018-12-06:
change your home location in MP as well to get similar results. looks like its set to bottom left at the moment.

anestetico, 2018-12-06:
[image]
[image]
[image]
[image]

meee1, 2018-12-06:
the only one you didn't try was home? which is what I suggested. just make sure its set correctly in flight planner first.

meee1, 2018-12-06:
also, you can drag the red marker around in the grid screen to tweak as well.

anestetico, 2018-12-06:
I tried home before but the home was inside the polygon, I put a fake home outside and it worked, I didn't knew it was like that, so how the "point" option works?
thank you very much, beautiful software
[image]
[image]

Naterater, 2018-12-06:
> 2\. the extra points are needed depending on what mode you use (servo/digicam). The reason they are useful is it causes the navigation controller to pull tighter to the start of the survey area,, [...]

Sorry to go back to this topic, but the straight-through waypoints don't do any more to get on-line because the L1 controller controls the maximum approach angle. They don't bring it any tighter to the survey area. It's kind of a bug for ardupilot, but I'll have to do some flight testing or illustrations to show you what I mean.

Well done on the rest of the improvements. I agree, this survey grid manager is a beautiful piece of software that has excellent options and very clear definitions without any assumptions.

meee1, 2018-12-09:
@Naterater please show with tests. pretty sure im correct.
im closing this issue, as it address's the issue originally shown.

closed 2018-12-09 by meee1