ArduPilot/MissionPlanner #1687: Issues with MissionPlanner-1.3.50 latest

IronDome2, 2017-11-22:
Issues with MissionPlanner-1.3.50 geotagging injects wrong coordinates into the images. The preview graphic shows the correct location, but resulting KML, TXT, CSV etc are all wrong coordinates see ample below of the tagged txt file output. The area is located in East Africa so clearly those coordinates are wrong.

Thanks in advance.

DSC04220.JPG -31077865 322473932 1401.75 359.22 4.96 1.17 99.5599975585938
DSC04221.JPG -31075030 322472968 1402.23999023438 354.69 5.74 -0.5 99.5099945068359
DSC04222.JPG -31072242 322471957 1401.40002441406 354.38 3.49 0.69 101.470001220703
DSC04223.JPG -31069362 322470928 1401.23999023438 354.25 5.86 -0.49 101.32999420166
DSC04224.JPG -31066486 322469849 1400.59997558594 353.88 5.6 2.7 101.169998168945
DSC04225.JPG -31063593 322468829 1400.4599609375 353.29 5.66 1.06 101.470001220703
DSC04226.JPG -31060692 322467821 1400.44995117188 357.05 7.07 2.31 102.040000915527
DSC04227.JPG -31057812 322466884 1400.669921875 354.62 5.55 -1.29 104.209999084473
DSC04228.JPG -31054913 322465862 1400.64001464844 350.61 5.81 -0.13 104.32999420166

IronDome2, 2017-11-26:
I just stopped using TLOGs now using dataflash logs (BIN)

meee1, 2017-12-05:
please try the latest beta. there was an issue with reading exif data

closed 2017-12-05 by meee1

IronDome2, 2017-12-05:
Thanks, you r absolute genius! I will try that.