ArduPilot/MissionPlanner #1690: MP opens up a myriad of Radius windows when it starts

expntly, 2017-11-26:
#### Issue details
When MP starts and before I connect to any vehicle, it spawns anywhere between 3 to 10 of these Radius windows:

[image]

Is this a bug, or is there a way I can disable this?

#### Version
1.3.49

rmackay9, 2017-11-27:
I think it's a bug.  I've seen it as well.

n1kitap, 2017-11-30:
Noted this as well at 1.3.49 and still get it in latest stable version.

meee1, 2017-12-05:
fixed here f9d4bffb

closed 2017-12-05 by meee1

expntly, 2017-12-08:
I just updated to 1.3.50, and saw the windows again.  Is that in this version?

[image]

expntly, 2018-01-03:
Any update?  Thanks a lot!

meee1, 2018-01-03:
update your MP

wisnufireball, 2018-03-20:
I just update my mission planner to 1.3.53 and a bug akin to this appear. It appear on Arduplane with sonar just after it completed the "getting param" process. And it won't stop spawning even after I disconnect the FC. APM:copter with sonar is fine, and Arduplane without sonar is also fine. I haven't test another firmware though.

System spec:
Mission planner 1.3.53 on Windows 10 64bit
Arduplane 3.4.8
APM:Copter 3.5.5
FC is Pixhawk, and sonar is Maxsonar EZ4 on analog mode.

meee1, 2018-03-20:
can you please posts 1 of your logs?

wisnufireball, 2018-03-21:
This is the logfile; https://drive.google.com/open?id=1etf7aV9YOCv19H7aXyuvs5x99pPLkDEg
Even when the Sonar hardware is disconnected, the windows is still spawning.