ArduPilot/MissionPlanner #1691: Rover Basic Tuning page update for ver 3.2

rmackay9, 2017-11-27:
We are getting close to releasing Rover 3.2.  This update comes with a number of parameter changes so I've created a mock-up of what the "Basic Tuning" page could look like to support the release.

[image]

rmackay9, 2017-11-27:
Ah, I see I got the parameter names (shown in red in the image above) reversed for NAVL1_DAMPING and NAVL1_PERIOD.

NickNunno, 2017-11-27:
Tuning currently requires flipping through multiple pages so getting these consolidated would be absolutely phenomenal! 

Also, I honestly cant recall off the top of my head but ensuring the basic tuning page is the primary landing page(or the tab holds the last page accessed) so the user can F key between Flight Plan and Tuning is/would be super helpful.

meee1, 2017-11-27:
tip @nicknanno double click basic tuning while you are in this screen, ti will pop out.

NickNunno, 2017-11-27:
Ah! Very Nice! Thanks!

rmackay9, 2017-12-09:
I'm going to list some issues I've found when testing this page:

- "Turn G MAX" only allows whole units (i.e. "1" or "2") but the increment for the parameter (in the description anyway) is 0.1.
- most of "Steering Rate" and "Speed 2 Throttle" gains are only showing 1 decimal (we should have 3).
- in the "Steering Mode" box the string is "Turn" but it should be "Turn Radius"

meee1, 2017-12-09:
@rmackay9  whats it set to in the param docs? it appears to be spelt wrong in the doc

https://github.com/ArduPilot/ardupilot/blob/master/libraries/APM_Control/AR_AttitudeControl.cpp#L52vs
https://github.com/ArduPilot/ardupilot/blob/master/libraries/APM_Control/AR_AttitudeControl.cpp#L59

extra E

meee1, 2017-12-09:
https://github.com/ArduPilot/ardupilot/pull/7382

closed 2017-12-12 by meee1

rmackay9, 2017-12-12:
Thanks @meee1!

commit 87dc057e (2017-12-12): ConfigArduRover: fix #1691