ArduPilot/MissionPlanner #1696: Manual configuration?

Jmelz, 2017-12-06:
#### Issue details
I cannot configure my plane because I'm using mission planner with a dragonlink and vector . I'm looking for a way to skip the wizard and manually enter that I have a plane and not a quad.
#### Version
Any
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Finwing penguin
#### Hardware type
_What autopilot hardware was used? 
Eagletree vector and Dragonlink

#### Logs
N/A

meee1, 2017-12-12:
the telemetry link has noting to do with the firmware loaded on your autopilot. please refer to the wiki regarding loading the correct firmware.

closed 2017-12-12 by meee1

Jmelz, 2017-12-12:
meee1 - I think I wasn't clear then. I am not at all trying to load firmware. I am simply trying to configure the software to display a plane with a few parameters I wish to manually tell it - so that it can display mavlink data from my dragonlink sourced by eagletree vector. I see no need that I should go through the wizard, when it does not apply.