ArduPilot/MissionPlanner #1698: MP doesn't connect when there is more than one vehicle on a link

olliw42, 2017-12-10:
#### Issue details
With e.g. a STorM32 gimbal controller, configured to emit Mavlink heartbeats with sysID = 71 and componentID = 67, and connected to a Pixhawk/ArduPilot, and Mission Planner connected to the Pixhawk, when Mission Planner aborts any CONNECT attempts with "Only 1 Heartbeat received". During the ca. 30s for which it tries to CONNECT, the HUD etc. is working, so it seems to have connected to at least one vehicle. 

With the STorM32 configured as a component of the autopilot, i.e. sysID = 1 and componentID = 67, then things work nicely.

I could be an issue with ArduPilots mavlink routing of heartbeats, but since it's happening for arducopter 3.6 dev and ardupilot-solo 1.5.3, I guess it's MissionPlanner.

#### Versions & Hardware type
MP 1.3.50, 1.3.51

solo cube with arducopter 3.6 dev
solo cube with ardupilot-solo 1.5.3
pixracer with arducopter 3.6 dev

meee1, 2017-12-10:
do you have a tlog, that should be recording everything seen by MP

olliw42, 2017-12-10:
I've never used tlogs so far
hopefully it's this: www.olliw.eu/drop/apm/2017-12-10-08-54-51.zip
here I tried to connect through a solo stock cube with solo-1.5.3 flashed and the PC connected to the cube's usb

thx for looking into this :)

meee1, 2017-12-10:
given the firmware you are running is the stock solo firmware, I doubt it will ever work correctly.

olliw42, 2017-12-10:
i) this is exactly so also with arducopter 3.6 dev, various hardwares
ii) at the time of that "old" firmware (I think 3.3 or 3.4), ca 2.5 years back, it actually used to work. Let's say like so: There once was a time where "everthing" with MissionPlanner, ArduCopter, and STorM32 was working together. Several things have changed in between, and we llost this state.

olliw42, 2017-12-10:
here is someone who seems to reported similar issues: https://discuss.ardupilot.org/t/storm32-gimbal-camera-control-and-problems-connecting-to-mp-when-gimbal-connected/23874

meee1, 2017-12-10:
ok so I looked at the log.
why is the gimbal spamming the port with command_ack's?

meee1, 2017-12-10:
2017-12-10T23:34:16.263 | FE | 3 | 0 | 0 | E9 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
2017-12-10T23:34:16.279 | FE | 3 | 0 | 0 | EA | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.310 | FE | 3 | 0 | 0 | EB | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.326 | FE | 3 | 0 | 0 | EC | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.341 | FE | 3 | 0 | 0 | ED | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.357 | FE | 3 | 0 | 0 | EE | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.388 | FE | 3 | 0 | 0 | EF | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.404 | FE | 3 | 0 | 0 | F0 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.419 | FE | 3 | 0 | 0 | F1 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.435 | FE | 3 | 0 | 0 | F2 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.466 | FE | 3 | 0 | 0 | F3 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.482 | FE | 3 | 0 | 0 | F4 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.497 | FE | 3 | 0 | 0 | F5 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.529 | FE | 3 | 0 | 0 | F6 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.544 | FE | 3 | 0 | 0 | F7 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.560 | FE | 3 | 0 | 0 | F8 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.575 | FE | 3 | 0 | 0 | F9 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.607 | FE | 3 | 0 | 0 | FA | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.622 | FE | 3 | 0 | 0 | FB | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.638 | FE | 3 | 0 | 0 | FC | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.669 | FE | 3 | 0 | 0 | FD | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.685 | FE | 3 | 0 | 0 | FE | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.700 | FE | 3 | 0 | 0 | FF | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.716 | FE | 3 | 0 | 0 | 0 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.747 | FE | 3 | 0 | 0 | 1 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.763 | FE | 3 | 0 | 0 | 2 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.778 | FE | 3 | 0 | 0 | 3 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.809 | FE | 3 | 0 | 0 | 4 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.809 | FE | 9 | 0 | 0 | 5 | 47 | 43 | 0 | mavlink_heartbeat_t | custom_mode | 6 | type | 26 | autopilot | 83 | base_mode | 0
2017-12-10T23:34:16.825 | FE | 3 | 0 | 0 | 6 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.841 | FE | 3 | 0 | 0 | 7 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.856 | FE | 3 | 0 | 0 | 8 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.887 | FE | 3 | 0 | 0 | 9 | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.903 | FE | 3 | 0 | 0 | A | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.919 | FE | 3 | 0 | 0 | B | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.950 | FE | 3 | 0 | 0 | C | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11
2017-12-10T23:34:16.965 | FE | 3 | 0 | 0 | D | 47 | 43 | 4D | mavlink_command_ack_t | command | 205 | result | 0 |   | sig | Len | 11

olliw42, 2017-12-11:
because it is receiving command_lomg messages, and is supposed to respond to each with an ACK?

(at least this once was the behavior required by the standard)

meee1, 2017-12-11:
the reason I ask that is its not mission planner doing it. so must be the pixhawk

olliw42, 2017-12-11:
yes, that is correct
pixhawk is sending the cmd_long-mount_control messages to the STorM32, which responds accordingly

I would, in some way, agree that it could be seen as an issue of the "new" mavlink routing rules used by ardupilot, which it uses since some time. However, this was my earlier argument, this happens also with arducopter-solo, which uses the "old" mavvlink routing rules, and for which it used to work.

meee1, 2017-12-11:
The long you provided shows no parameter request originating form mission planner. so I don't think this log even shows the original problem.

can you post a new tlog that shows the issue.

I expect to see MP requesting params from both ArduPilot and the gimbal. with the gimbal failing.

olliw42, 2017-12-11:
ähm ... I think that this is what is happening
this log should be complete from the moment I've powered up both the STorM32 and the pixhawk!
that is, I've powered up both, waited for the pixhawk beep tones which indicate that the pixhawk is ready for connect, then tried to connected, waited the ca 30 secs, and zipped the tlog after having seen that its size doesn't increase anymore ... I wouldn't know what else I could/should do ...

meee1, 2017-12-11:
thanks for the follow up @olliw42 found the underlying issue. has been there for a while.

meee1, 2017-12-11:
to test the Master build.
make sure you are on 1.3.51 and goto help, hold down control, and click update beta
this should update you to master

closed 2017-12-11 by meee1

olliw42, 2017-12-11:
have tested it a couple of times for both a pixracer with 3.6dev and a solo stock cube with ardupilot-solo-1.5.3 ... didn't get any failures ... so, seems to work!

THX!

yes, the issue had been there for a while ... it's just that it only now became relevant for me ;)

BTW: there is another (minor) issue which is also there since a while, and which I reported some time back: The combo box field below the two UART fields and next to the CONNECT bottom, where the current vehicle is displayed, is NOT updated correctly upon a connect. I can e.g. create this situation:
- start Mission Planner => vehicle combobox is "empty"
- connect to a link with two vehicles, e.g. QUADROTOR-1 and GIMBAL-71 => MP connects, but vehicle combobox is still empty
- select GIMBAL-71 => vehicle combobox shows GIMBAL-71
- disconnect
- press connect again => MP connects, and vehicle combobox shows GIMBAL-71

however, in (essentially) all cases the HUD etc. pp is actually that of QUADROTOR-1, and not of that of GIMBAL-71, in contrast to what the vehicle combobox tells!

can be quite confusing sometimes

thx, Olli

meee1, 2017-12-11:
ok ive fixed the update of the current mav as well here 
https://github.com/ArduPilot/MissionPlanner/commit/ccba3f6b655c35c6dbc099d94a45e807a534a347

olliw42, 2017-12-12:
fantastic!!

thx so much :)

commit 9e1b9667 (2017-12-11): MAVLinkInterface: fix #1698 connect issue