ArduPilot/MissionPlanner #1700: Suggestion: Highlight checked fields in data field selection dialogs

olliw42, 2017-12-10:
#### Issue details
In the flight data screen, with "Tuning" checked a plot opens showing data in real time. It's curves can be selected by double clicking opening a dialog window, with all available data field for selection.

I like this tool VERY much, but generally have troubles to see which fields are checked. It's really a pure visual problem. It nearly always takes me quite some time of concentrated looking to get it right.

It thus would be great if the selected fields could be made more easily visible, e.g. by a darkgray background instead of black, or anything else which wouldn't be hard to code.

I note that similar dialogs/windows are used in other occasions too, I just don't have them in mind. The suggestion would apply to them too.

I agree that my ailing eyes are part of the problem ;)

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2017-12-12:
this use to happen, and is implemented. just need to check why it stopped working

olliw42, 2018-02-01:
THANK you soo much for resolving this, really extremely helpful !

closed 2018-02-01 by olliw42