ArduPilot/MissionPlanner #1701: Rover: Radio calibration should not change SERVOx_REVERSED parameters

rmackay9, 2017-12-11:
In Rover-3.2.0 (and later versions) we've properly separated the input and output min/max/trim/reversed values.  Now the MP's Initial Setup >> Mandatory Hardware >> RC Calibration screen's "reversed" check boxes should only affect should only affect the RCx_REVERSED parameters (i.e. input) not the output (SERVOx_REVERSED).

Here is a sample report where the user is intending to modify the RC input reversing and getting into trouble with the outputs also being reversed. https://discuss.ardupilot.org/t/rc-configuration/23821/9

rmackay9, 2017-12-11:
It may actually be better to completely remove these reverse check boxes for Rover as we have done for Copter and then ask the users to perform any input reversing in their transmitter.

The reason I think the check boxes is that users tend to get themselves into trouble by reversing the input when they really need to modify the output.

meee1, 2017-12-12:
ive added a new servo output screen. and removed the RC/SERVO reversed link

closed 2017-12-19 by meee1