ArduPilot/MissionPlanner #1707: guided mode not working for plane

tridge, 2017-12-16:
GUIDED mode is not working for plane at the moment. It does work for copter
It looks like there is a special case for plane here:
https://github.com/diydrones/MissionPlanner/blob/master/Mavlink/MAVLinkInterface.cs#L3048
plane and copter use the same interface for guided these days, so I think what is needed is to remove the special case.

meee1, 2017-12-16:
they don't use the same interface for guided, I though this was the case as well. however plane doesn't implement the fill new method. only the old method

plane supports altitude change only via the new method.

meee1, 2017-12-16:
plane https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/GCS_Mavlink.cpp#L1787

tridge, 2017-12-16:
hmm, I'm puzzled, the set target position code is here:
https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/GCS_Mavlink.cpp#L853
and set alt code is here:
https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/GCS_Mavlink.cpp#L875
both are supported in copter, here:
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L677

for copter, MP sends the set target pos command as expected:
https://github.com/ArduPilot/MissionPlanner/blob/master/Mavlink/MAVLinkInterface.cs#L3057
but for plane it does something quite different:
https://github.com/ArduPilot/MissionPlanner/blob/master/Mavlink/MAVLinkInterface.cs#L3050

I'm not sure what you mean by the "fill new" method?

rmackay9, 2017-12-16:
Maybe @meee1 means the Plane doesn't fully support the SET_POSITION_TARGET_LOCAL_NED message, it only the SET_POSITION_TARGET_GLOBAL_INT message?  Personally I view these two messages as the same in terms of new-ness.  The "old" method for me would be using the mission-item to execute commands in guided.. that was a crappy old way to do it.  COMMAND_LONG, COMMAND_INT and these two SET_POSITION_TARGET_xxx messages are the proper way to do this.

meee1, 2017-12-16:
The old method uses a version of the waypoint protocol.
the new method is  SET_POSITION_TARGET_GLOBAL_INT
copter supports both methods
plane only supports the old waypoint protocol method
the only support in plane for  SET_POSITION_TARGET_GLOBAL_INT is to change alt.

this is what MP has the code to use the old method for plane

tridge, 2017-12-16:
I'm happy to add SET_POSITION_TARGET_GLOBAL_INT of course, but the code MP has in for plane right now definitely doesn't work. It enters guided mode, but doesn't do the position set, so the plane just circles when you start guided.

meee1, 2017-12-16:
are you testing with beta or stable?

meee1, 2017-12-16:
I believe stable is broken, beta is working.

meee1, 2017-12-16:
I'm pushing a new stable MP atm

tridge, 2017-12-16:
thanks Michael! I've tested the new release and it works.

closed 2017-12-16 by tridge