ArduPilot/MissionPlanner #1717: Mode 6 is set to Manual without the users consent.

lthall, 2017-12-23:
#### Issue details
The final mode on Plane is automatically set to manual in Flight Mode 6. If this is changed and the flight mode screen is used to change any other mode then mode 6 is reset to manual.

This very nearly caused a crash as I normally set this to RTL. The aircraft did not react as I expected so I rushed to the mission planner and hit RTL on the Actions tab. When I examined what happened I found that the mode 6 was set to manual. I had changed a flight mode but forgotten that this feature/bug existed and accidentally set mode 6 to manual at the same time.

#### Version
All versions. The simulation options shows this nicely.

#### Platform
Plane

#### Airframe type
All that I am aware of.

#### Hardware type
All, including the simulator

#### Logs
Not needed.

rmackay9, 2017-12-23:
Perhaps @tridge needs to confirm but I think we should remove the MP's requirement that channel 6 be set to MANUAL.  This same "feature" was in place for Rover's flight mode setup screen within MP but we removed it recently.

meee1, 2017-12-23:
its a legacy apm 2 thing

lthall, 2017-12-23:
Thanks all!!