ArduPilot/MissionPlanner #1721: RTK Base always 1Hz

tuloski, 2018-01-09:
I see here: https://github.com/ArduPilot/MissionPlanner/blob/3814bb7584f9653d3a884ec8e0792b9563ae0e43/ExtLibs/Utilities/ubx_m8p.cs

that for the RTK, the base frequency is always set to 1Hz, which is bad according to ublox documentation. The base and rover should have the same rate and since the minimum rate in arducopter is 5Hz, there are some problems (i.e. not easy RTK fix). Can the rate of the base be selected with a parameter in the RTK/GPS inject window?

meee1, 2018-01-10:
for moving base yes it is bad, for a normal rtk base its is normal.

please provide a link to your source.

tuloski, 2018-01-11:
Yes, you're right. I didn't notice that the phrase: "Additionally, to ensure optimal performance the Base and Rover receiver should use the same navigation rate. " was related only to moving base mode.

So rover at 5Hz and base at 1Hz should be the best option for fixed base, while 4Hz rover and 4Hz base is the best for moving base.

Sorry for the false issue :).