ArduPilot/MissionPlanner #1727: RTK NETWORK Corrections

Naterater, 2018-01-16:
#### Issue details
Many RTK networks offer free network (not single baseline) corrections. Currently I can only access NTRIP corrections from one station at a time. Is there a way for mission planner to implement or process network corrections through some standard protocols? MAX and iMAX multi base station corrections would be useful when operating in the middle of many stations but not close to any.

#### Version
All

#### Platform
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type

#### Hardware type
Here+ without base station.

#### Logs

meee1, 2018-01-16:
do you have more info on this technology? I have seen NTRIP used with VRS stations.

ie a virtual base station.

Naterater, 2018-01-17:
I am only a user of the technology, and unfortunately I can only provide links. Using a networked correction requires sending the latitude and longitude of the rover to the network, according AgLeader. I use the Iowa RTN, and the only solution that I can get working through mission planner NTRIP is the Single Site RTCM 3.x solution. The Iowa RTN also offers more solutions in the second link. I have to use my personal login.

[](http://www.agleader.com/blog/ntrip-cors-how-it-works-and-what-is-involved/)
[](https://iowadot.gov/rtn/IaRTNRTKProducts-NTRIP-QuickGuide.pdf?ver=2017-09-22-094506-477)

RTNFriend, 2018-02-07:
Hi, I am trying to figure out why Mission Planner is not sending NMEA GGA messages to an RTK network via its built in NTRIP client. I posted my finding here (as Yuriy, dronehobby) https://discuss.ardupilot.org/t/rtk-gps-inject-with-ntrip/23925/11?u=dronehobby

But yeah, I am still far from the solution, and it would be great someone from development team to have a look at the issue. Thanks !

As a side note: some drones have RTK receivers on them which allows 1cm accurate positioning (for precise aerial mapping for example) but such receivers need Virtual or Real base GNSS raw data to work with (usually in RTCM 3.x format). 

We do rely on network solutions - the VRS "sations" because modelled GNSS raw  from them are are closer the the user compared to a raw data from a distant single station. Closer vectors allows quicker and more reliable RTK fixed solutions (compared to long vectors).

meee1, 2018-02-07:
please try beta MP. I did make changes with the formatting of the nmea string sent recently.

MP does send a GGA string. its just if the other end like the format im generating.

Naterater, 2018-02-08:
The beta MP works with VRS now. Thank you so much Michael! Hopefully this will be released into master sometime soon.

closed 2018-02-08 by Naterater

RTNFriend, 2018-02-08:
Tried to update MP to latest beta, Pressed and held down Ctrl button and clicked update button, the updater downloads stuff, but each components fails to install, will try to install on different machine.