ArduPilot/MissionPlanner #1741: Altitude type automatically changes from relative to absolute

nex-robotics, 2018-01-26:
If I write waypoints to Pixhawk and read them back... the altitude type automatically changes to Absolute.
I usually read mission after writing it... to be extra sure.
First time I encountered this.. I did not notice. Then in next mission, I edited the mission and loaded new waypoints not noticing the altitude type had changed to Absolute. The copter started flying with absolute altitude. Had to bring it back and land. Thankfully no casualties :-)

I am using Mission Planner version 1.3.45
Copter firmware 3.5.1
Airframe : Hexa
Hardware Pixhawk

ranqingfa, 2019-02-28:
Its a very serious problem, got a crash.

nex-robotics, 2019-02-28:
Seems to be resolved in Mission Planner version 1.3.59

nex-robotics, 2019-02-28:
> Its a very serious problem, got a crash.

Seems to be resolved in Mission PLanner 1.3.59
Can you check and confirm.
I do not know the cause and how it was resolved.

closed 2019-08-31 by nex-robotics