ArduPilot/MissionPlanner #1742: Change VTOL Avatar from Quadcopter to Plane or new VTOL graphic

iskess, 2018-01-27:
When MP identifies the aircraft as a VTOL it uses the Quadcopter graphic. Most VTOL aircraft look more like airplanes, spend most of their time as a Plane, and are based off the Arduplane code. The graphic should ideally be a new VTOL type, but in the meantime its better to display a plane. 

[image]

iskess, 2018-01-27:
Upon closer inspection I now see that MP thinks this is a Copter and NOT a Quadplane. All the Modes refer to Arducopter. This is rather confusing. I'm going to change Q_MAV_TYPE from 20 back to 0.

SSUAV, 2018-01-27:
It would be great If the graphic could toggle between the plane & quad depending what mode the quadplane is in.

iskess, 2018-01-28:
That’s a great idea. It serves the same purpose as my other issue #1309