ArduPilot/MissionPlanner #1744: Rover: RC calibration should say ".. and throttle is centered"

rmackay9, 2018-01-29:
Near the end of the RC calibration procedure, a messages box pops up saying, "Ensure all your sticks are centered and throttle is *down*, and click ok to continue" but for a rover, the throttle should be centered.

rmackay9, 2018-01-29:
Also as a side note, the MP is saving RCx_TRIM values but these values aren't displayed on the final screen.
[image]

ncx94, 2018-02-26:
I think it's more tricky than that. Some rovers don't have reverse and then will use full throttle range with 0 throttle at the bottom.

I don't remember right now if there is a parameter to enable reverse driving, but if yes, this parameter should be checked to control the correct display on the message :)