ArduPilot/MissionPlanner #1746: MP crashes when trying to display CTUN.DAlt of one particular .bin file

rmackay9, 2018-01-29:
There's a battery failsafe report in the Copter-3.5.5-rc1 beta testing thread.  If I attempt to open the .bin file provided, it opens but if I then attempt to graph CTUN.DAlt it crashes.

discussion: https://discuss.ardupilot.org/t/copter-3-5-5-rc1-available-for-beta-testing/25162/10
log file from @Iketh: https://www.dropbox.com/s/6o48ttcpchowrwc/2018-01-28%2014-22-12.bin?dl=0

[image]

meee1, 2018-01-29:
please update to latest beta MP.

rmackay9, 2018-01-29:
I'm using 1.3.2.8. build 1.3.52.8.. at this moment it seems like the latest version available but I'll try again in a couple of hours.

rmackay9, 2019-04-06:
haven't seen this problem again so must be fixed, thanks!

closed 2019-04-06 by rmackay9

meee1, 2019-04-06:
I fixed another issue in beta recently that caused an issue with a bad multiplier defined in a log.