ArduPilot/MissionPlanner #1747: FFT windows can not decode "batch DF logs" dataformat

amilcarlucas, 2018-01-29:
#### Issue details
The ACC2 and ACC3 data are not properly displayed in the FFT window.

#### Version
Arducopter master and MP 1.3.52.6

#### Platform
[  ] All
[  ] AntennaTracker
[ *] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
X8 (octa quad)

#### Hardware type
pixhawk

#### Logs
[Documents.zip](https://github.com/ArduPilot/MissionPlanner/files/1674320/Documents.zip)

meee1, 2018-01-29:
these logs don't contain that data

meee1, 2018-01-29:
there was a bug in the beta. now resolved

amilcarlucas, 2018-01-30:
Thanks for fixing this Michael!

amilcarlucas, 2018-01-30:
Sorry Michael, but I just updated to 1.3.52.8 beta and it pops up an error dialog. It still does not work :(

meee1, 2018-01-30:
please control-click the update beta button
should put you on 1.3.52.10

amilcarlucas, 2018-01-30:
[image]

That gives me this error message

meee1, 2018-01-30:
agh that's a beta update bug.
downgrade then upgrade