ArduPilot/MissionPlanner #1748: Mission Planner "Makeover"

Naterater, 2018-01-30:
#### Issue details
According to ~80% the community (of 103 users following ardupilot.org on facebook), Mission Planner could use a makeover. To begin, MP is packed with a wonderful amount of features. Maybe those features have outgrown the current organization of mission planner. While there are many lists of "feature requests" here already, I don't think this is why people think MP needs a makeover. To clarify, this is not a request for addition or reduction in features.

I believe the two most useful things for a mission planner makeover would be
- Restructuring of the upper tabs and creating a ribbon containing most of the commands and buttons
- Reorganization of the display space into windows that are movable and adaptable

I think this is the right first step to making mission planner more user-friendly and adaptable to users of all levels. It may or may not seem like a big change, but most users are very familiar with the "ribbon" layout and managing windows. By placing almost all commands and options at the top in a ribbon that can be hidden, the entire lower screen can fit more data applicable to the user. (This would effectively move the buttons under the HUD to the top ribbon)

I envision the top ribbon containing ~9 tabs
- Setup
- Mission Planning
- Pre-Flight
- Flight (MAIN)
- Tuning
- Analysis
- Development
- Help
- Donate

Within each ribbon tab, there will be 4 main types of buttons.
- Open a new pop-up window over everything else (think firmware upgrade or wizard)
- Open a new window within the main display area below-but not allowed to overlap (HUD/Live Map/Quick Data Grid)
- Basic - does something (connect/disconnect, takeoff, preflight calibration, write WP, etc.)
- Drop-down/expandable lists (NEW - Quickly set flight modes via checkboxes)

Since almost everything would be located in the top ribbon that is collapsible, there is more room for displaying whatever the user wants during flight monitoring. The command buttons could be selected while still viewing the "quick data grid." All features would be in the ribbon and not hidden behind left clicks or double-clicks.

I understand this is not a paid endeavor, but I really think the community needs this. This is the "face" of ardupilot.

jaimemachuca, 2018-01-31:
This research on the UX/UI by Intel is very interesting and valuable. http://juca.work/work/drones/

Naterater, 2018-12-22:
It doesn't appear to me that there will be much effort on these big-picture items. Here is a list of smaller, simpler tasks that could make a difference.

#2027
#2028
#2032
#2038
#2050
#2013
#2026
#2030
#2031
#1860
#2029

Maybe after these are addressed/discussed, larger architectural changes could be made.