ArduPilot/MissionPlanner #1753: Feature Request - Altimeter and Speed settings on HUD

RowanWatkins, 2018-02-03:
#### Issue details

Due to the standards adopted by different countries, the reference datum for altitude and speed settings in aviation varies. It will greatly enhance the functionality of Mission Planner to be able to customize the display to show the users preferred units of measurement.

In some countries the standard used for altitude is feet, distance is in nautical miles and vertical speed in feet per min. In metric airspace the defaults are the metric equivalent.

UAV operators appear to prefer values that don't interface with the standards used by other aviation users in the same environment. Most times this doesn't present an issue as the UAV is operating below the other airspace users and well clear of any traffic. This is not always the case, and UAV operators need to know the industry standards and be able to interface on that basis.

For professional UAV operations the ability to communicate with other aircraft and ATC using the same units of measurement is paramount to safe operations and a prerequisite for being taken seriously in the aviation community. To make Mission Planner a more effective tool for all operations it would be beneficial to be able to display the units of measurement appropriate to the environment where the UAV is operating.

For traffic awareness and separation purposes you may need to report your position, altitude and speed to an aircraft. That aircraft may be using feet based on QNH as his altitude reference, speed in knots and distance in Nautical Miles. It does not work to tell him your altitude in meters, speed in M/S and distance in Kilometers.

Most of these suggested changes should be easily implemented as the information is already available it is just a case of an extra conversion table to allow separate values for Horizontal and Vertical Distance with the addition of the Altimeter settings.

#### Version
All

#### Platform
[  ] AntennaTracker
[  ] Copter

#### Airframe type
All

HUD Features

[image]

Unit Selections

[image]

Altimeter Datum Selection

[image]

SSUAV, 2018-02-03:
+1
Excellent

Naterater, 2018-02-14:
I really like the idea! I struggle with the granularity when the units are in m/s. Right now the value is rounded and jumps between whole number values and the green cruise line is hard to see. The newer airspeed sensors smooth the data a lot better than in the past, and sometimes a jump of 1 m/s is almost a 10% increase in airspeed. Maybe the airspeed indicator could use a decimal place or a scrolling unit like full scale HUD's do.
[image]

jace26, 2019-01-03:
This looks great, I assume the altimeter setting would be adjustable and affect the actual baro offset, or else it would be pretty pointless.