ArduPilot/MissionPlanner #1754: SiK radio no longer working with 1.3.50.0 or 1.3.52.0

timvand, 2018-02-04:
#### Issue details
_Please describe the problem, or desired feature_
Attempted to connect to radio on COM3, which gernerated connection error.  Updated driver, fail.  Remove driver, reboot, add driver via \mission planner\Driver\DPInstx64.exe tool, attempt to connect via COM3, fail.
#### Version
_What version was the issue encountered with_
1.3.50.0 or 1.3.52.0
#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Plane
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk 1, SiK radio
#### Logs
_Please provide a link to any relevant logs that show the issue_
N/A

[image]

meee1, 2018-02-04:
That error message means something else is using the port.
I have verified here, and nothing wrong with missionplanner

timvand, 2018-02-04:
Does it require being run as admin?  When the radio is unplugged from the USB port, the COM port connection is removed.

timvand, 2018-02-04:
How much do I need to pay to help resolve the issue?

meee1, 2018-02-04:
depending on the model of radio you are using it most likely has a ftdi chipset inside it. I use rfd900 and sik radios on the same install all the time.
you need to close other apps on your pc to get to the bottom of what is opening the port and not closing it.

meee1, 2018-02-04:
also check device manager to ensure you have the correct port
[image]

timvand, 2018-02-04:
Doesn't removing the device from the USB port on the PC remove the port in use?  In my experience it does, as I see the COM port being added and removed in the computer management\device manager window when the radio is added and removed from the USB port.

meee1, 2018-02-04:
not always, reboot the pc as well.

timvand, 2018-02-04:
Been doing reboots as well.

No radio plugged in.
[image]

433 radio added
[image]

COM15 detail
[image]

Connection fail
[image]

timvand, 2018-02-04:
So, do a reboot, and then add the radio?

meee1, 2018-02-04:
Could be a driver issue as well.

MP has never shipped with Silicon Labs drivers.

timvand, 2018-02-04:
Shit.

What drivers are to be used?

timvand, 2018-02-05:
And here is a similar issue, which the download of the SL drivers appeared to resolve the issue.
https://discuss.ardupilot.org/t/new-dell-pc-with-windows-10-home-mp-doesnt-see-the-serial-ports-and-crashes/22724

closed 2018-02-09 by meee1