ArduPilot/MissionPlanner #1755: Mavlink: regenerate.bat not working

Katawann, 2018-02-06:
#### Issue details
I would like to regenerate Mavlink.cs because I added a new Mavlink message, but the regenerate.bat is not working. I also tried only using the mavgen but the paths are not set properly.

#### Version
Mission Planner 1.3.52.10
Windows 10

#### Logs
Using Cygwin or the command prompt, execute regenerate.bat gives the following output:

```
$ ./regenerate.bat

D:\Documents\GitHub\MissionPlanner\ExtLibs\Mavlink>rd /s /q "mavlink"

D:\Documents\GitHub\MissionPlanner\ExtLibs\Mavlink>python -m pymavlink.tools.mavgen --lang=csharp --wire-protocol=2.0 "message_definitions\ardupilotmega.xml"
'python' is not recognized as an internal or external command,
operable program or batch file.

D:\Documents\GitHub\MissionPlanner\ExtLibs\Mavlink>copy /y "mavlink\ardupilotmega\mavlink.cs" "Mavlink.cs"
The system cannot find the path specified.

D:\Documents\GitHub\MissionPlanner\ExtLibs\Mavlink>pause
Press any key to continue . . .
```

Only with mavgen and common.xml: 

```
$ python -m pymavlink.tools.mavgen --lang=csharp --wire-protocol=2.0 "message_definitions\common.xml"
Validating message_definitions\common.xml
Parsing message_definitions\common.xml
[ .. ]
Note about the message ID
[ .. ]
Found 158 MAVLink message types in 1 XML files
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/cygdrive/d/Documents/GitHub/MissionPlanner/ExtLibs/Mavlink/pymavlink/tools/mavgen.py", line 30, in <module>
    mavgen.mavgen(args, args.definitions)
  File "pymavlink/generator/mavgen.py", line 116, in mavgen
    mavgen_csharp.generate(opts.output, xml)
  File "pymavlink/generator/mavgen_csharp.py", line 389, in generate
    os.makedirs(directory)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 2] No such file or directory: 'mavlink/message_definitions\\common'
```

magicrub, 2018-02-06:
This isn't your problem but I thought I'd mention anyway that Mission planner will need the ArduPilotmega.xml to work properly, it will automatically pull in common.xml.

meee1, 2018-02-06:
install python for windows. and run the bat file from windows, not cygwin

Katawann, 2018-02-06:
@magicrub yes I had some problems with that but from the ArduPilot side with mavproxy

@meee1 This works! I also forgot to add Python to my PATH, but now the file is generated. Thanks a lot for your quick answers !

closed 2018-02-06 by Katawann

magicrub, 2018-02-06:
@meee1 maybe its' worth to do a couple checks in the bat file to throw some errors like this pseudo-code:
if PATH != contain("Python") then
  echo "Please install Python"
  exit
else if RUNNING_CYGWIN ==true then
  "Please run from Windows command prompt"
  exit

sediq-khan, 2019-03-16:
Hi @magicrub ,

I still run into this problem, attached please see the error screenshot.

[image]