ArduPilot/MissionPlanner #1764: TECS data is missing in log browser

expntly, 2018-02-19:
#### Issue details
For some reason I don't see the "TECS" category anymore in the Log Browser (DF logs).  The TECS data is definitely in the log, as I can tell from mavlogdump.py and also by using the type filter in MP (3rd column), which does show me TECS lines.

#### Version
1.3.52.0

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Sub

#### Airframe type
QuadPlane X8

#### Hardware type
PH2.1

magicrub, 2018-02-20:
What ArduPlane firmware version?

meee1, 2018-02-20:
please provide the bin file. but i asume its not in the log. the header/filter will apear because it shows all FMT types

expntly, 2018-02-21:
@magicrub 3.8.2

@meee1 : What do you mean it's not in the log?  

> $ mavlogdump.py --t TECS data.BIN | head
> 2018-02-21 11:53:17.49: TECS {TimeUS : 522992754, h : -0.0863990858197, dh : 0.00260599097237, hdem : -0.0613990798593, dhdem : 0.250000059605, spdem : 28.6837749481, sp : 3.0, dsp : 1.60659217834, ith : 0.00847248267382, iph : 283.489440918, th : 0.928472459316, ph : -0.240032449365, dspdem : 0.0, w : 1.0, f : 0}
> 2018-02-21 11:53:17.70: TECS {TimeUS : 523195613, h : -0.0812967643142, dh : 0.00894750561565, hdem : 0.0586634241045, dhdem : 0.713124990463, spdem : 28.6837749481, sp : 4.498503685, dsp : 1.66962397099, ith : 0.0255213938653, iph : 275.339080811, th : 0.98609316349, ph : -0.244800463319, dspdem : 0.0, w : 1.0, f : 0}
> 2018-02-21 11:53:17.80: TECS {TimeUS : 523296412, h : -0.0782966539264, dh : 0.0253399591893, hdem : 0.151410296559, dhdem : 0.927468657494, spdem : 28.6837749481, sp : 5.15444517136, dsp : 1.56844115257, ith : 0.0339194200933, iph : 271.349029541, th : 1.0, ph : -0.171905443072, dspdem : 0.0, w : 1.0, f : 0}
> 2018-02-21 11:53:17.90: TECS {TimeUS : 523398372, h : -0.0748888477683, dh : -0.013683738187, hdem : 0.264519810677, dhdem : 1.13109517097, spdem : 28.6837749481, sp : 5.73723649979, dsp : 1.5912541151, ith : 0.0423696562648, iph : 267.35635376, th : 1.0, ph : -0.118298709393, dspdem : 0.0, w : 1.0, f : 0}
> 2018-02-21 11:53:18.10: TECS {TimeUS : 523601267, h : -0.06642113626, dh : -0.00126859359443, hdem : 0.547805190086, dhdem : 1.50831341743, spdem : 28.6837749481, sp : 6.41242265701, dsp : 1.47189700603, ith : 0.059128664434, iph : 259.51864624, th : 1.0, ph : -0.0493814274669, dspdem : 0.0, w : 1.0, f : 0}
> 2018-02-21 11:53:18.30: TECS {TimeUS : 523803796, h : -0.0678204894066, dh : 0.00826169550419, hdem : 0.90097028017, dhdem : 1.84875309467, spdem : 28.6837749481, sp : 6.71663808823, dsp : 1.4810988903, ith : 0.0758869349957, iph : 251.814025879, th : 1.0, ph : -0.00129635853227, dspdem : 0.0, w : 1.0, f : 0}
> ...

meee1, 2018-02-22:
please provide the log file

expntly, 2018-05-24:
@meee1  log file here: https://drive.google.com/open?id=1pFmOK65QeDeqO2pJwrclpZZShzoMEn8m

And screenshot to show TECS, TECS2, QTUN messages, while these same categories are missing from the grouping on the right hand side, leaving me with no way to plot them.  This is ArduPlane 3.8.5beta1, and a quadplane.

[image]

meee1, 2018-05-24:
@expntly  MP version number?

meee1, 2018-05-25:
I just pushed a fix here https://github.com/ArduPilot/MissionPlanner/commit/e536d2046fee6d08ba4970d43892829dc081001c

expntly, 2018-05-25:
Thanks.  When are you releasing just so I know when to check?

meee1, 2018-05-25:
im create a new beta release now, so give it 30 mins. and then Help > update beta

expntly, 2018-06-08:
I just did an update (not beta) to `1.3.56 build 1.3.6672.30.243`, and I'm seeing the same problem with `RATE` being missing.  Did your change make it into this release or is it another bug?