ArduPilot/MissionPlanner #1773: TLog Playback Broken for Multiple Component IDs.

edwinhayes, 2018-03-05:
#### Issue details
Playback of telemetry logs (recorded by Mission Planner) which contain messages from more than one component ID seems to fail.  The failure mode is that during playback, nothing visible happens, even though the playback progress bar advances as expected.  While playback of the raw file fails, if we then run the file through an external utility which removes all the MAVLink messages not coming from cID 1, then the resulting file plays as expected, so it seems to be the presence of some message from another MAVLink component in the file is the cause of the issue.  Not sure whether it's a specific message, or just the presence of messages from other components in general.

#### Version
1.3.4 and 1.3.52 are affected, haven't tested anything before 1.3.4.

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Quad

#### Hardware type
Pixhawk V2.1

#### Logs
Can't share the original log due to commercial sensitivity.  Should be straight-forward to replicate though.

meee1, 2018-03-05:
need to clarify.

components or sysids or both?

edwinhayes, 2018-03-05:
Ah... good question.  Seems like there are some messages from another sID in the log file, which I wouldn't normally expect to be there.  So possible this is a screw up on our side - I will have this investigated.

edwinhayes, 2018-03-05:
I'm told that filtering by sID==1 (so you have multiple cIDs, but only one sID) still results in the abnormal playback behaviour, so just differences in cID alone is sufficient to replicate the problem.  I guess it's possible that it's just a specific message ID which is responsible; I'll see if we can narrow it down further.

meee1, 2018-03-05:
I test with multiple sysid's and the same compid fairly often.

but don't really test against different compid's on the same sysid