ArduPilot/MissionPlanner #1774: Missions: draw loiter tangent exits from loiter radius instead of center

magicrub, 2018-03-06:
When drawing a mission with a loiter point that has the tangent exit flag set, it would be nice to actually draw the tangent exit path!

[image]

Explanation of the problem:
https://youtu.be/XuSl9Io93aQ?t=1709

iskess, 2018-03-06:
As displayed in your example, the loiter radius doesn’t always match the WP radius, giving a false sense of the aircraft’s planned position. Would you also propose that the dashed circles be drawn to match the planned radius at loiter points? This would display an accurate tangent exit path like you are seeking.

magicrub, 2018-03-06:
@iskess yes, that would be nice too.