ArduPilot/MissionPlanner #1778: Mission Planner 1.3.53.0 Verify Height bug

dvorw, 2018-03-14:
#### Issue details
Using Verify Height and Relative Altitude, Mission Planner plans using Absolute Altitude values.
If Verify Height is unchecked, the plan is created for a fixed Relative Altitude, as expected.
Verify Height + Absolute also behaves as expected. But Verify Height + Relative gives the same plan.

#### Version
1.3.52.0

#### Platform

I have Mission Planner installed on 3 different platforms (2 desktop PCs and a laptop), all running Windows 10 (possibly different builds), with Spanish system language. 

This error occurs on the PCs, but not on the laptop. 

I am planning missions without connection to the drone. On the two PCs, Mission Planner has never been connected to the drone, nor set up for a particular firmware.  The laptop has been used with the drone in the past.

meee1, 2018-03-20:
im going to take a step back and ask this.

what is it you think verify hight option does?

dvorw, 2018-03-21:
Hi

My understanding and use of Verify Height:

Adjusts ‘constant’ uav altitude value in mission to account for differences in terrain elevation.

In Relative mode this altitude is referenced wrt height above the ground (Home). In Absolute mode, is expressed as altitude above sea level.

dvorw, 2018-03-23:
Hello

I found the source of the problem - when Home Altitude is incorrect.