ArduPilot/MissionPlanner #1782: Change simulation AP firmware version

expntly, 2018-03-19:
#### Issue details
I couldn't find a way to change the ArduPilot firmware version when running the Simulation.  This is more a feature request to be able to select: master, previous release, etc.

#### Version
1.3.52

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

magicrub, 2018-03-19:
What do you mean "change the version"? There's a version.h file in each vehicle type, is that what you meant? https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/version.h

meee1, 2018-03-19:
currently the only silt firmware compiled is master.
I will look into providing stable as well.

charlesacknin, 2018-03-20:
I mean be able to select a minimum of BETA, LATEST STABLE, MASTER (arducopter/arduplane release) that SITL uses.

Basically the same way you can select the ArduPilot firmware version when you install a new pixhawk/etc.

Thanks!

expntly, 2018-03-20:
I mean be able to select a minimum of BETA, LATEST STABLE, MASTER (arducopter/arduplane release) that SITL uses.

Basically the same way you can select the ArduPilot firmware version when you install a new pixhawk/etc.

Thanks!

abe-mart, 2019-03-08:
Any update on this one?

omriorion, 2020-01-29:
Has anything developed regarding this request?