ArduPilot/MissionPlanner #1783: double simulation using Mission planner

chrismdn, 2018-03-20:
#### Issue details
Hi everyone, I am not sure whether I am posting in the right section.
I was wondering if it would be possible to run a double simulation using mission planner. 
here is my thought; I'd like to set a couple of waypoints on mission planner, have my quadcopter/plane following those waypoints on X-Plane 10 (using the SITL) and at the same time I would like to see the real quadcopter flying in the "real airspace"( following the same waypoints set on mission planner.
#### Version
_What version was the issue encountered with_
1.3.53
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
quadcopter/fixed wing

#### Hardware type
 pixhawk,

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2018-03-20:
which one is following which?

just connect to sitl, connect to the real drone, and use MP's follow me mode.

although I do think this is a bad idea.

chrismdn, 2018-03-21:
Hi, 

Thanks for your reply 
May I know why do you think it is a bad idea?