ArduPilot/MissionPlanner #1784: Nuget Package Issue (System.Drawing.Common)

alphajafarabadi, 2018-03-20:
I clone the **least Mission Planner source code** using **Visual Studio** then i try to **Rebuild** it but i can't cause of project decencies. **Nuget** restores all packages successfully except _**System.Drawing.Common** 4.5.0-preview2-26225-02_, in Nuget package manager the last version of System.Drawing.Common is 4.5.0-preview1-26216-02.

Are you use from a custom/third-party Nuget source?

meee1, 2018-03-20:
please do a 
nuget.exe sources Add -Name "myget" -Source https://dotnet.myget.org/F/dotnet-core/api/v3/index.json

closed 2018-03-20 by meee1

alphajafarabadi, 2018-03-21:
Thank you @meee1 😄 my problem has been solved but now i have a new one...
Please check attached file.

Details:
Windows 10 1709
Visual Studio Enterprise 15.6.3 Full
.NET Core/Standard/Framework all versions
[image]