ArduPilot/MissionPlanner #1785: Add other vector format support using the OGR library

everydayduffy, 2018-03-20:
#### Issue details
Is it possible to add the option to load a polygon from vector formats other than shapefiles please? OGR drivers support many formats, e.g. GeoPackage (see: http://www.gdal.org/ogr_formats.html). It would be a nice addition to be able to use these.  

#### Version
Any

Many thanks

meee1, 2018-03-20:
yes its possible.

what format did you have in mind?
as an overlay or waypoints?

everydayduffy, 2018-03-21:
The latest QGIS now uses [GeoPackage](http://www.geopackage.org/) as a default format for vector data, moving away from shapefiles. Thats the format I had in mind since i'm starting to adopt it as well (only one system file per vector and no limit on column name character length among other benefits).  

My first thought was being able to import polygons as an overlay, but as you mentioned waypoints, that would also be a useful feature.

meee1, 2018-03-21:
MP already has GDAL support. do you have a sample file I could test with?

everydayduffy, 2018-03-21:
Attached a sample geopackage polygon.

[testgpkg.zip](https://github.com/ArduPilot/MissionPlanner/files/1832638/testgpkg.zip)

kikislater, 2018-03-23:
+1 for adding others formats support

meee1, 2018-03-28:
ive added gpkg support, currently miss labeled though as KML Overlay

flightplanner > right click > Map tools > kml overlay