ArduPilot/MissionPlanner #1790: Mission Planner with Other Base Stations

Fukusshima, 2018-03-27:
Hi,

We can not see the precision levels with Here + RTK Kit. Because it uses only one frequency.

So we would like to use the Saptentrio m2 UAS card as a rover, Stonex S9 III Receiver (or any dual frequency Receiver) base station. The Saptentrio card can be used as a Rover, but for Stonex, the mission planner can not obtain the location from the NMEA data coming from the base station. I've examined the NMEA data types sent from Here + and Stonex, but both send GN, GS, GP data, but the sending speeds can be different. Stonex supports maximum to 1 Hz speed. I set the sending speed of data types like GGA, GLL... at 1 HZ but Mission Planner can't read again. Is it due to the sending speed? I have no resources.

Mission Planner Version is 1.3.52.

Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

PixHawk 2.

Regards.

meee1, 2018-03-27:
NMEA data is not used as part of a base station.
the base station needs to output rtcm v3, base and obs messages. and needs to support base station functions

Fukusshima, 2018-03-28:
We don't expect like a quick reply from PixHawk Generator. Thank you for reply Michael. 

I have worked for ARI Defence and Aviation(http://arisavunma.com.tr/en) which expetcly prodeus UAV sistems.

Execuse me I couldn't understand what is the base and obs messages. What is their content? Would you please explain me? 

Best Regards.

jmachuca77, 2018-03-28:
The septentrio NR series bases, for example, can be setup to output RTCM v3 correction messages over TCP, you can then use mission planner to receive these corrections and forward them to the m2 rover receiver.
Your stonex receiver specs say it can output RTCM corrections so there should be a way to do the same with it.

Best regards,

Jaime Machuca
CEO | Element Aero | www.element.aero
Mail. jaime@element.aero

Off. +52 (33) 3805 4627
Cel. +52 1 (33) 3945 3350

Este correo electrónico y cualquier archivo transmitido en él, son confidenciales y para uso exclusivo de los individuos y entidades a quienes está dirigido. Si usted no es el destinatario previsto o la persona encargada de recibirlo, y tiene por error este mensaje, queda prohibido y sin validez el uso, difusión, re-envío, reimpresión o copia. Toda oferta y/o aceptación de propuestas comerciales, celebración de contratos u otros actos tendientes a la adquisición de bienes o servicios, así como el establecimiento de cualquier clase de obligación legal para Element Aero SAPI de CV, deberá confirmarse por escrito firmado autógrafamente por funcionario competente, excepto que se cuente con un contrato vigente que autorice el uso de este medio para tales fines. Si usted recibió este correo por equivocación, favor de notificar inmediatamente por este medio a su remitente, y después borrarlo de su correo