ArduPilot/MissionPlanner #1791: Add CSV Export to selected Parameters in logged graph

lordneeko, 2018-03-28:
#### Issue details
I need to be able to easily correlate data from my logs. That's not possible when viewing graph data on two vastly different scales.  For example, Compass heading vs RSSI. I want to be able to export those two parameters to CSV, pull them into a tool (like excel) and perform correlation analysis on them.

#### Version
1.3.53

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
skyviper

meee1, 2018-03-28:
are you doing a DF log? you can  graph on the left and right axis using left and right clicks

lordneeko, 2018-03-28:
Ok well that's info I didn't know lol.  But still I can't do a solid correlation analysis using the graph. I would still like to be able to export to csv if possible

meee1, 2018-03-28:
DF or tlog?

DF is a csv already
Tlog, there is an option in the mavlink log graph to export as csv

lordneeko, 2018-03-28:
Well I have .bin files from mine which are not csv, they are binary.  I'm
not sure the exact format..it's the files logged on the skyviper sd card.

meee1, 2018-03-28:
[image]
use the convert bin to log file option. then open the log file in excel.

magicrub, 2018-03-28:
@lordneeko in addition to have a left/right click to offer two graph plot scales, you can also double-click a given variable in a log and it will pop up a dialog box to adjust an offset and/or scaler. It won't change the plot, it just applies the offset/scale to the existing plot scale

lordneeko, 2018-03-28:
Ok I'll try these things thanks. Was that documented somewhere and I missed
it?