ArduPilot/MissionPlanner #1793: Feature Request: read mission form .bin file directly

magicrub, 2018-03-28:
I often need to know a mission from a bin. Right not it takes three steps:
- press button "create KML+gpx" (which also creates waypoint files)
- sort through multiple waypoint files to discard the junk partial ones via notepad (see https://github.com/ArduPilot/MissionPlanner/issues/1718)
- open the rest of them one by one and see if any of them look like they match a flown mission

It would be nice if the Flight Plan page had a "load bin file" button (or was combined in existing button) that would allow selecting which of the multiple missions I want to view. That is, I'd like to load a bin and then have it offer mission 1,2,3,4,5, and then I can select 4.

meee1, 2018-03-28:
so add an interface to view each mission within that file? and provide extraction of that single mission

magicrub, 2018-03-29:
yes. If a single mission is found then just load it. Else, I'm picturing a dialog box that says "the log contains more than one mission, which would you like to load?" with a list. not sure what the list would show, maybe the waypoint length count?

Also, it would be nice if the existing "Load WF File" button simply accepted an additional Type: (*.txt; *.waypoints; *.shp; *.plan; ***.bin**)

magicrub, 2018-03-29:
Oh, I know something that could be listed in addition to waypoint count: the time delta from when it was in the log since boot. That would help alleviate the pain from dealing with https://github.com/ArduPilot/MissionPlanner/issues/1718. If I see that the logs are just a few seconds apart then I can disregard them.