ArduPilot/MissionPlanner #1796: TradHeli Setup Page

ChristopherOlson, 2018-03-30:
The setup page for TradHeli is out of date and doesn't work properly. @bnsgeyer and I are considering requesting it's removal from MP and force users to use the Full Parameter List for heli setup. There is no doubt the issues with MP's heli setup page could be fixed, if identified. However, with heli now supporting three different frame types, and a forth coming soon, it will be a very difficult task to keep it up to date and working with the rapid pace of heli development.

The Full Parameter List "just works" and we have been recommending using only it for awhile now for Trad Heli.

Comments?

rmackay9, 2018-03-30:
Personally I like having custom setup screens for the vehicles.  So we recently improved [MP's setup screen for Rover](https://github.com/ArduPilot/MissionPlanner/issues/1691) and I think it works well.

Still, it's really up to you guys.

ChristopherOlson, 2018-03-31:
Right now we have a problem with users attempting to use the heli setup page. And one recent case it applied the totally wrong throttle mode and severely messed up the fellow's setup, disabling his throttle control for his heli.  Helicopters are not as "generic" as multicopters, and for instance once I PR the changes to support the TT swashplates it causes a further complication with how do you support this from a GUI-type interface?  And then we have tandem rotor machines with either Chinook-style layout, or SyncroCopter layout, or coaxial head layout, we have dual swashplates and setup for each type is totally different. They now don't even have a tail servo, but in the case of a coaxial the fourth servo is used for differential collective.  Where the Chinook-style or SyncroCopter using mixing for differential collective.

I suppose we could try to do something for the conventional single rotor machines.  But even the full parameter tree does not support the settings for heli's as MP tries to be too helpful and limit them to settings for multicopters - returns a "bad value" message.

So we are still trying figure out what to do about it, and how to go about fixing it if it is decided the current page is to try to be saved.  I don't own a Windows computer to try to do any development or testing for it. Although I do have it running on linux with mono, but it is not fully functional.

lvale, 2018-07-31:
@meee1 This is getting a bit serious with 3.6. We can try to work out a revamp of the page, if you want.

ChristopherOlson, 2018-07-31:
If it is to be revamped I would request a similar look and feel of the helicopter setup page in QGC. But the major issue we have right now is that people on 3.6 are trying to use this page in MP to set parameters that don't even exist. The page "reads" settings from parameters that do exist and applies them to parameters that don't exist, writing new values to them in the background. And it breaks the setup of the machine.

The last time this page worked was AC3.3.3 and it was not fully functional then. The changes from 3.3 to 3.6 are so significant, even with other things like the SERVO library, that if it is to be saved pretty much need to start over from scratch on it. But it needs to be disabled or removed pronto in the mean time as it is breaking people's setups on 3.6.

lvale, 2018-08-09:
@meee1 Do you have a estimate to this ?