ArduPilot/MissionPlanner #1802: Changing MP measurement units doesn't change altitude value for mission

suraqis, 2018-04-02:
#### Issue details
After you input WPs in Flight Plan view and then changed altitude units (meters to feet or vice versa) altitude values for already input WPs will not convert. It's better to convert those altitudes, to avoid the need to correct every WP manually. Also if you wrote the same WP in meters and then in feet it will result in different altitudes and may cause the crash. 

#### Version
MP 1.3.55

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

Ryanf55, 2020-06-06:
How are you changing it in Flight Plan view? Flight plan does not not show the units of altitude, or really any of the data.

 In [documentation](https://ardupilot.org/copter/docs/mission-command-list.html#waypoint), it says "Alt - the target altitude above home in meters. " 

[image]

pbreed, 2022-06-04:
Doc says this is in meters, but if you have ft selected as planner units display is actually in ft.
So it should 
a)Either match the docs, or the docs should change.
b)Ideally it should show the units  on EVERYTHING that changes .
So Radius, Alt, Dist all should show units in grid or item labels.