ArduPilot/MissionPlanner #1804: "Abort landing" button do nothing when land in VTOL mode

suraqis, 2018-04-03:
#### Issue details
When landing in VTOL mode "Abort Landing" button from "Actions" pane does nothing. There's no normal way to cancel VTOL landing immediately from MP.

#### Version
MP 1.3.55
AP 3.9.0-dev

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Quadplane

#### Hardware type
SITL

#### Logs
Easily reproducible in SITL.

suraqis, 2018-04-04:
I think it should stop descending an climb to altitude where landing descend started in VTOL mode. More simple is to go QLOITER. Of course it should be supported in AP.

I see this feature usefull when there’s some obstacle or people at the place of landing.

Awilonk, 2018-12-30:
@suraqis agree!
I am tring to add a button "abort VTOL land" to do so. Just switch to QLOITER mode and simulate a joystick input to hold the vehicle's altitude and position. It's a dummy solution, could give me some sugestions? @magicrub .