ArduPilot/MissionPlanner #1805: MP 1.3.55 Flight Plan "Add Below" Bugged

thatsnailguy, 2018-04-04:
#### Issue details
When "Add Below" button is clicked in Flight Plan error occurs.

#### Version
1.3.55 6660.28588

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
N/A

#### Hardware type
N/A

#### Logs

An error has occurred
System.FormatException: Input string was not in a correct format.
   at System.Number.ParseDouble(String value, NumberStyles options, NumberFormatInfo
numfmt)
   at System.Double.Parse(String s)
   at MissionPlanner.GCSViews.FlightPlanner.writeKML() in C:\MP\GCSViews\FlightPlanner.cs:line
1295
   at MissionPlanner.GCSViews.FlightPlanner.BUT_Add_Click(Object sender, EventArgs e)
in C:\MP\GCSViews\FlightPlanner.cs:line 877
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at MissionPlanner.Controls.MyButton.OnClick(EventArgs e) in C:\MP\ExtLibs\Controls\MyButton.cs:line
157
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at MissionPlanner.Controls.MyButton.OnMouseUp(MouseEventArgs mevent) in C:\MP\ExtLibs\Controls\MyButton.cs:line
186
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32
clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam,
IntPtr lparam)

Report this Error???

arunnarenthiran, 2018-04-09:
I have been facing the same error when I click 'Accept' after planning the flight in 'Create survey grid'. Have you been able to solve it?
[image]

meee1, 2018-04-09:
please try 1.3.56