ArduPilot/MissionPlanner #1820: Pixhawk doesn't recgonise two rangefinders (sonar: Maxbotix I2CXL EZ4) as collision avoidance.

deepanshidrone, 2018-04-25:
#### Issue details
I'm currently trying to set two sonars (Maxbotix I2CXL EZ4) as rangefinders for collision avoidance in a QUADCOPTER. I am obviously using an I2C splitter. First of all, when seting the parameters in the Parameters list, how does the Pixhawk know which sonar is the RNGFND_1 and which is the RNGFND_2 ? When I have a look to the rangefinder log, the first one appears but the second one seems to don't be recognized by the Pixhawk, what is the problem? 

#### Version
Mission planner 1.3.56

#### Platform
Copter

#### Airframe type
Quadcopter
#### Hardware type
Pixhawk
#### Logs
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2018-04-26:
This is really a flight code issue so it should go over in the ArduPilot flight code issues list instead of here.

wiki setup guide for Copter-3.5 is here: http://ardupilot.org/copter/docs/common-rangefinder-objectavoidance.html

I suspect the issue is that there is an I2C address conflict (i.e. both range finders are using the same I2C address) and/or the orientation parameter hasn't been set for one or both of the lidar.  ArduPilot also only supports a single rangefinder in a particular direction.  So for example, we don't support two range finders both pointing forward.  if one is forward-left and the other forward-right that should work though.

closed 2018-04-26 by rmackay9

deepanshidrone, 2018-04-26:
Thank you so much for your time! 
The orientation parameter is already set different for both sonars, so I guess it is an I2C address problem. How can I fix that?