ArduPilot/MissionPlanner #1836: RTK FIX or FLOAT is not able to reach by GPS even after Fixed LLA in Survey in

siddyyagno3, 2018-05-14:
#### Issue details
_Please describe the problem, or desired feature_
Please provide help to check if RTCM message is being transmitted through telemetry or if it some other issue? GPS is still 3D fix even after letting it run for 20 mins after reaching Survey in. Accuracy Set to 0.9.
Does it support SBAS configuration?

####Version
_What version was the issue encountered with_
Copter V3.5.5 MP 1.356 build

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk 2 cube with Here+ Rtk GPS Base and Rover 
Telemetry MRO radio

#### Logs
_Please provide a link to any relevant logs that show the issue_
[image]
[image]

geofrancis, 2018-05-15:
uncheck the "inject MSG Type" box

siddyyagno3, 2018-05-15:
Unchecking the inject msg did not helped. Tried increasing the accuracy to 2m and using usb instead of telemetry radio. But still the issue prevails.

geofrancis, 2018-05-15:
have you updated the gps firmware ?

siddyyagno3, 2018-05-15:
@geofrancis yup tested both 1.30 and 1.40

geofrancis, 2018-05-15:
its a configuration issue, even with barely any sats it should go to float almost immediatly.

siddyyagno3, 2018-05-15:
I also thought so I firstly set the configuration to default and finally I removed 1.30 and installed 1.40 but in vain.

siddyyagno3, 2018-05-15:
Another thing I tried installing PX4 firmware but GPS rover is not sat count is zero blue light blink there but when I roll back to ardupilot there is 3D lock but no float or fixed. Yellow lights blink on gps.

siddyyagno3, 2018-05-24:
The issue was with the carrier board of pixhawk 2 cube gps port. Getting float on older pixhawk.

closed 2018-05-24 by siddyyagno3