ArduPilot/MissionPlanner #1839: BUG - Terrain following drop down box crazyness.

JacksonUAS, 2018-05-21:
#### Issue details
Terrain following drop down box disappears on connection to UAV. 

Toggling the theme brings it back (either connected to UAV or not)

#### Version
1.3.56

meee1, 2018-05-21:
what firmware version on the vechile?

JacksonUAS, 2018-05-21:
Test with AC 3.4.2 and 3.5.3, no change.

meee1, 2018-05-21:
are you connecting via the normal way?

in the status tab (flightdata, bottom left), what shows under version?

JacksonUAS, 2018-05-21:
Via UDP and usb com port - same results.

We had the behaviour on two separate PCs yesterday, and today the issue wont come back with nothing obviously changed...

meee1, 2018-05-21:
During connect MP requests the firmware version on the autopilot. if for some reason this fails, menu items get disabled as they are not supported.

so my guess its to do with packet loss

JacksonUAS, 2018-05-22:
Did some more testing, and yeah it seems to be a combination of packet loss andthe version message being unpopulated  and then when that version message returns mission planner doesnt redisplay the drop down box on reconnect until the theme is changed.