ArduPilot/MissionPlanner #1841: [BUG]Mission Planner crashes on start/stop of Video input from Easycap USBTV007

Shackmeister, 2018-05-21:
Hi People

Im experiencing a crash when starting or stopping the Video input.
If selected Via HUD screen, right click, Start camera, it starts correctly as long as no video device is selected.
If I select my video device and press start, I get an error message "Camera Fail: xxx" (Ill post the rest later), and Mission planner crashes instantly.

The Frame grabber used is an Easycap, based on the UTV007 chip.
Driver used is: [OEMDrv.inf.txt](https://github.com/ArduPilot/MissionPlanner/files/2023913/OEMDrv.inf.txt)

Mission planner used is 1.3.56

Please let me know if there is any other information I should provide.

Thank you for the work

Shackmeister, 2018-05-22:
a GIF to show the main problem:
Error message says: "Camera Fail: a device, attached to the system, is not working correctly".
[image]

Shackmeister, 2018-05-22:
If I then try to open the device via HUD -> right click -> Start Camera I get a fault again, and MP crashes
[image]

Shackmeister, 2018-05-22:
but weirdly I can get it running by restarting my computer and run the framegrabber via HUD -> right click -> Start Camera. But even at this point I need to be careful with changing settings, since it might easily cause a crash
[image]

Shackmeister, 2018-05-22:
@meee1 Hi Michael
First, thanks for this wonderful piece of software! you have truly done an amazing job :)
second, is there any debug info I could provide to make it easier to track down?

meee1, 2018-05-22:
I don't think there is much I can do.
your first error "a device, attached to the system, is not working correctly" is a windows system error. indicating there is something wrong with the device or cable.

i would try finding other drivers that work or similar.

Shackmeister, 2018-05-22:
even if it's a windows problem, it would be very nice if MP didnt crash fully when the problems occur

meee1, 2018-05-22:
check the windows event log for the application crash. no logging MP does will catch a crash at this level.

Shackmeister, 2018-05-22:
@meee1 here is what Event Viewer gives me:
Application: MissionPlanner.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.AccessViolationException
   at WebCamService.Capture+NativeMethods.CopyMemory(IntPtr, IntPtr, Int32)
   at WebCamService.Capture.DirectShowLib.ISampleGrabberCB.BufferCB(Double, IntPtr, Int32)

and

Faulting application name: MissionPlanner.exe, version: 1.0.0.0, time stamp: 0xdfc63617
Faulting module name: ntdll.dll, version: 10.0.17134.1, time stamp: 0x207580e2
Exception code: 0xc0000005
Fault offset: 0x00000000000a0b2c
Faulting process id: 0x2f90
Faulting application start time: 0x01d3f1b64ad12e50
Faulting application path: C:\Program Files (x86)\Mission Planner\MissionPlanner.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: 1a4492d3-07cf-4c05-8808-c8db57b93a9b
Faulting package full name: 
Faulting package-relative application ID:

meee1, 2018-05-22:
I found something that might cause one of those errors. im uploading a new beta now

Shackmeister, 2018-05-22:
Sounds great! Should I compile it myself to test or can I find a nightly/beta build somewhere?

meee1, 2018-05-22:
MP help screen > check for updates. wait another 10 before you try though