ArduPilot/MissionPlanner #1849: RC Calibration Wizard Improvement

Naterater, 2018-06-05:
#### Issue details
There are way too many users that end up with inputs (and consequently outputs) reversed. This stems from incorrect RC Calibration user errors. Although we say in the docs to make sure that the green bars move with the sticks, there are still far too many people that don't get it right or don't notice. It results in too many crashes, and this problem could be solved with a better step-by-step wizard and calibrating one thing at a time.

I have heard that other GCS's do it differently. I haven't used them, but I think the following procedure would make the most sense:

1. Confirm the user wants to calibrate his/her radio.
2. Ask user to remove any radio trims, center sticks, lower throttle, and "press any key"
3. Ask user to roll right only, hold, and "press any key" like the accelerometer calibration
4. Ask user to roll left only, hold, and "press any key"
5. Ask user to pitch nose-up, hold, and "press any key"
6. Ask user to pitch nose-down...
7. Throttle up...
8. Yaw right...
9. Yaw left...
10. Flip all of the mode switches and dials (and record like the whole procedure is recorded currently)
11. Return all sticks to neutral positions

Sure, the "corners" of the stick positions may not be 100% accurately recorded for min/max, but there should be no problem if an RC input is beyond the max or min; it will still be full-value. We're more concerned here about users not getting to full value due to not calibrating.

By using each individual step, the reversals and RCmap channels can be automatically calculated. It also accomodates users who are used to a reversed stick.

Behind the scenes:
1. Confirm calibration intent.
2. Confirm neutral start & set trims.
3. Find CH with the most change, set RCMAP_ROLL to that channel, if change is negative then reverse the channel, record RC(CH)_MAX (or MIN if reversed)
4. Record RC(CH)_MIN (or MAX if reversed)
5. Find CH with the most change, set RCMAP_PITCH to that channel, if change is negative then reverse the channel, record RC(CH)_MAX (or MIN if reversed)
6. Record RC(CH)_MIN (or MAX if reversed)
7. Find CH with the most change, set RCMAP_THROTTLE to that channel, if change is negative then reverse the channel, record RC(CH)_MAX (or MIN if reversed).
8. Find CH with the most change, set RCMAP_YAW to that channel, if change is negative then reverse the channel, record RC(CH)_MAX (or MIN if reversed)
9. Record RC(CH)_MIN (or MAX if reversed)
10. Find all of the rest of the Min's and Max's by using the old technique
11. Set RC(n)_TRIM for all channels, and RC(throttle)_MIN.

#### Version
All

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
ALL

#### Hardware type
Anything with a transmitter

#### Logs
N/A

rmackay9, 2018-06-06:
Thanks @Naterater for raising this issue.

A number of rover users have bumped into this so reducing the support burden would be great.  I think if possible it would be nice to avoid the user needing to press, "OK" so many times but whatever works I guess.  Perhaps a button that says, "set reverse" and then it asks the user to move the roll stick right and then it just detects that the pwm channel has changed and sets the reversed parameter accordingly.

Naterater, 2018-12-01:
Support requests are still popping up due to double-reversed inputs and outputs that aren't recognized in plane. This is the easiest way to solve those problems and avoid crashes for first-time users.

Ryanf55, 2024-05-08:
Wiki PR for the time being:
https://github.com/ArduPilot/ardupilot_wiki/pull/5943

My brother missed this one on setup and I broke some props. 
It's not intuitive (yet).