ArduPilot/MissionPlanner #1852: DShot throttle unavailable

Naterater, 2018-06-10:
#### Issue details
I fly planes and need to monitor throttle for plane health. CH3_Percent does not work for DShot ESC's

#### Version
1.3.56.4

Plane 3.9 beta 1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Twin engine electric V-Tail

#### Hardware type
Cube, RFD900, BLHeli_32 ESC's

#### Logs

Naterater, 2018-12-01:
Throttle percentages are important to view. Still want this feature.

meee1, 2018-12-01:
do you have a log showing this?

Naterater, 2018-12-01:
https://www.dropbox.com/s/pho5rwo1bbh62uz/2018-11-30%2015-32-25.tlog?dl=0

And I'm talking about having something similar to CH3_percent but for other channels (because DSHOT ESC's have to be on Aux Channels.