ArduPilot/MissionPlanner #1853: RC Calibration allows trims below RC_Min

Naterater, 2018-06-10:
#### Issue details
RC calibration does not set trim for channels that aren't used during calibration. When migrating from an old parameter set that had the trim set low, the calibration doesn't reset that trim to be between min and max. Results in a pre-arm failure.

#### Version
1.3.56.4

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Sub

#### Hardware type
FRsky X9d+ and X8r

#### Logs
_Please provide a link to any relevant logs that show the issue_

WickedShell, 2018-06-11:
Either MIN/MAX/TRIM should all be set, or none should be set. In this case if it never got a valid input during calibration not setting the channel is the appropriate thing to do.

rmackay9, 2018-06-11:
Perhaps the same issue or a related issue.  It seems that RCx_TRIM is being set while RCx_MIN/MAX are not being set.

I've seen this on at least two boards/vehicles.  One was a new board that was just being setup, the other was an existing board.  https://github.com/ArduPilot/ardupilot/issues/8622

Naterater, 2018-06-11:
Calibrating should at least provide a sanity check on RC parameters. That's the point of running the RC calibration. It doesn't matter if a channel is active or not.

If there is an RC input error causing a pre-arm failure, most people would recommend and think to run an RC calibration. If that RC calibration does nothing, users are in an endless loop without manually modifying parameters. That shouldn't be a primary method used to troubleshoot when a calibration procedure is already in place.

Maybe the resolution should be to set inactive channels to 1100, 1500, and 1900 for Min, Trim, and Max.

https://github.com/ArduPilot/ardupilot/issues/8622

rmackay9, 2018-07-06:
This came up again in the Copter-3.6.0-rc3/-rc4 testing thread for brandon mcdougal..

rmackay9, 2018-07-23:
This has come up for Plane users now and is leading to @tridge writing work around in AP.
https://github.com/ArduPilot/ardupilot/pull/8982

rmackay9, 2018-07-23:
@meee1 tells me that this is now fixed in both the stable and beta MP releases.

meee1, 2018-07-23:
was fixed here https://github.com/ArduPilot/MissionPlanner/commit/b5c51608d027f99a0123c71195e55126b9604d61

closed 2018-12-01 by Naterater

arkanme, 2019-02-15:
where i find this settings pls help me i,am looking that from last 3 h and dont find RCx_MIN/MAX

skystarUK, 2019-04-11:
You say the issue was fixed, but I just upgraded from arducopter v3.4 to v3.6.7 and was met with a virtually all rcX_trims being wrong.
It’s kind of annoying because the rcX_option is set to 0 “do nothing” so the warning is pointless

I did a radio recalibration to see if the problem was caused by me not calibrating for a long time. The result was it actually messed up the settings again that I had previously manually corrected in the parameters list.