ArduPilot/MissionPlanner #1861: WP_RADIUS does not save in Flight Plan screen.

GrahamDyer, 2018-06-17:
#### Issue details
On the Flight Plan screen, if one changes the WP Radius, then leave the screen and return, WP Radius reverts to the previous value. And parameter WP_RADIUS does not change. 

[image]

Only if the actual parameter is changed in the Full Parameter List does the WP Radius value change stick on the Flight Plan screen

#### Version
V1.3.56

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Twin boom plane

#### Hardware type
AUAV-X2 & Pixhawk

#### Logs

meee1, 2018-06-17:
agreed, by design, its not until you upload a mission is it set.

this has always been the case

GrahamDyer, 2018-06-17:
That seems unintuitive, e.g, I set up and fly a mission, then decide to change the waypoint radius because the plane only turns _after_ hitting the waypoints, but I can't set it HERE even though it says that is the value I need to change. I have to go find the parameter in the Full Parameter List and change it there. , That doesn't make sense, having the value there is either superfluous AND misleading OR it should work correctly.