ArduPilot/MissionPlanner #1868: auto grid survey waypoints are changed after klicking to accept. Why?

Ralfde, 2018-06-24:
#### Issue details
When I set a survey grid wit auto survey grid, it does show the waypoints correct, but if I klick to accept, the waypoints are set different:

#### Version
Mission Planner 1.3.56 build 1.3.6672.30243

#### Platform
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad X

I want this waypoints:
[image]

, but after klickint to accept, I get this waypoints. Why?
[image]

GrahamDyer, 2018-06-26:
Try set the altitude to 50m or higher?

Ralfde, 2018-06-26:
tried 5m altitude. It happens only if all Overshot and Leadin are set to 0m:
[image]

Ralfde, 2018-06-26:
tried 5m altitude. It happens only if all Overshot and Leadin are set to 0m: