ArduPilot/MissionPlanner #1872: 3D visualization of planning and flight possible?

GrahamDyer, 2018-06-26:
#### Issue details
[image]

The AirRails GCS has an amazing feature where one can view the flight and do the planning in 3D.

Isn't that wonderful, to visualize both in a 3D environment? Is this possible to implement in MP?

There is a video here: https://www.youtube.com/watch?v=LT1lDytQOi4

#### Version
All

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
All
#### Hardware type
All

#### Logs
N/A

GrahamDyer, 2018-07-04:
Yes, but that's for post-planning or post-flight outside of Mission Planner. You can also upload a KML to http://doarama.com/ for an amazing after-flight visualization.

Having 3D inside Mission Planner is the next big thing.

kikislater, 2018-07-05:
After flight, kml is available !
Nice Cesium usage !

MyOnlyOnlySunshine, 2018-08-03:
Hope to join the ArcGIS 3D map