ArduPilot/MissionPlanner #1882: Two output file names resolved to the same output path

YandanYang, 2018-07-07:
I use Visual Studio 2017 to build Mission Planner. But I got some errors:
"Two output file names resolved to the same output path:"obj\Debug\netstandard2.0\MissionPlanner.Strings.resources"."
Has anyone met this kind  of problem before?
I tried to  copy and rename the wrong windows, but new error window came out later.
And I have never used C# before. Can anyone help me?

wuxiangchao, 2018-07-19:
我也遇到了同样的问题，弄了半天还是没找到解决办法。

YandanYang, 2018-07-23:
It is the problem of the way you download it. I at first downloaded it as .zip.
But my friend download this project with git, and everything works well. 
I tried git too. Problem fixed.

YandanYang, 2018-07-23:
“Github source-code zip files cannot be used because these do not contain the appropriate submodule source code. You must use git!”

meee1, 2018-07-23:
please use windows 10 to compile. windows 7 fails because of the language code does not exist

zpalgy, 2018-08-07:
I experienced the same problem on W7 . Moving the solution to W10 solved it.

kaomoneus, 2018-12-10:
I think Mission Planner owners should fix their build instructions, where it is necessary to state that we able to build it only under win 10. Otherwise I would consider this as a blocker issue.