ArduPilot/MissionPlanner #1886: Drop-down list of board choices for firmware loading

tridge, 2018-07-10:
The current "auto-detect and fallback to wizard" method of selecting the right firmware for a board is not working well. There are several problems with it:
- the auto-detection can't cope with boards where there are multiple possible firmwares (eg. px4-v3 and fmuv3)
- the wizard asks questions like "is this is pixhawk", but what is the answer for a cube? Some users have answered yes and ended up with incorrect compass orientations etc

I think a better solution would be a drop-down list of all possible boards, with the auto-detection code just giving the default value. This means the user can immediately see all the possible selections.
This is particularly important now we're adding separate firmwares for Pixhawk4, PH4-mini, CUAVv5, CubeBlack etc. One of the best ways of users starting to become aware of what firmware they should use would be this drop-down list.
 
ping @rmackay9

OXINARF, 2018-07-12:
I think this is a good option and I'd like to add that choosing the board should be done before choosing the vehicle type as for some boards we only build some vehicle types (Disco, Bebop and SkyViper come to mind).

rmackay9, 2019-05-04:
I hope with the Manifest file now available from AP that we can implement this.  Certainly on the support forums there are quite often postings by people having troubles getting the right firmware.

For example, [Keith Downes reports on the latest Copter-3.6.9-rc2 beta testing thread that he can't get the NuttX version for some reason](https://discuss.ardupilot.org/t/copter-3-6-9-rc2-available-for-beta-testing/41756/12).

bys1123, 2019-05-04:
Sounds good.