ArduPilot/MissionPlanner #1888: Unable to access debug messages on telnet 2300

Kelly-Foster, 2018-07-11:
@OXINARF provided numerous suggestions for getting access to debug printf() messages on port 2300 in a telnet window.  That doesn't appear to be working.  Please advise.  Happy to augment wiki once flow is proven / understood.

Kelly-Foster, 2018-07-11:
I restarted MP with debug console output, and now I am seeing few diagnostic messages in putty telnet port 2300 window.  Is it necessary to have the console output enable, or just coincidental in my system somehow?  I would like to update wiki on this capability, and don't want to assume.

meee1, 2018-07-11:
plaintext messages are output to the console and port 2300.
mavlink status messages are only output to the console.

OXINARF, 2018-07-12:
@Kelly-Foster You shouldn't need to have console output enabled to get the messages in telnet. Can you try disabling it and seeing if you get messages there or not?